patmo141/odc_public #1: error in blender

b4zz4, 2016-02-06:
no module named odc

patmo141, 2016-02-08:
can you post the error message from the Blender Console or a screen shot of the pop up.  It will tell me what line the error is in.

haquedmd, 2016-02-13:
The implant module seems to be working great. @patmo141 : Do you usually stay in this window when planning or creating surgical guides for dental implants? Do you plan on doing any additional videos for idiot rookies (like me)? 

[image]

patmo141, 2016-02-14:
This issue is closed.  Starting a new issue related to implant module documentation

debaetsr, 2016-02-23:
I think you missed classes.py line 41 and 42

patmo141, 2016-02-24:
Looks like it.  Strange that replace/find didn't get that.

patmo141, 2017-03-29:
fixed

commit 49e0833e (2016-02-13): old references to odc instead of odc_public fixes #1
closed 2017-03-29 by patmo141