patmo141/odc_public #4: Setup issue

Guisan, 2016-03-05:
Help me Patrick!  
Windows 10 
Blender 2.76b
Install-Easy

Report: Error
Traceback (most recent call last):
   File "C:\Program Files\Blender Foundation\Blender\2.76\scripts\modules\addon_utils.py", line 343, in  enable mod.register()
  File "C:\Users\BOMBEI\AppData\Roaming\Blender Foundation\Blender\2.76\scripts\addons\odc_public-master\__init__.py", line 203, in register import classes, odcutils, crown, margin, bridge, splint, implant, panel, help #, crown, implant, splint, panel, odcmenus, bgl_utils
  File "C:\Users\BOMBEI\AppData\Roaming\Blender Foundation\Blender\2.76\scripts\addons\classes.py", line 15, in <module> import odc_public
ImportError: No module named 'odc_public'
Tanks
Guido

patmo141, 2016-03-05:
Hi Guido,

You may need to delete the old addon, restart blender and reinstall.  The line your error references (line 15 in classes.py) no longer has that import.

Guisan, 2016-03-06:
Hi Patrick,
All right! It works!
Thank you very much.

closed 2016-03-07 by patmo141