patmo141/odc_public #7: Splints : KeyError: 'bpy_prop_collection[key]:key "odc_public" not found'

Leeda611, 2016-03-09:
Hi, 
I tried several times designing a splint..(even tried a very small splint just on the occlusal as a simple test) but the same error always appears.  
Would appreciate some guidance on how to correct this please.
Thank you
Lee
[image]

patmo141, 2016-03-09:
Thanks for the big report.  Indeed a few references have slipped by my sear/replace.  I will fix this evening when I get home.  If you search for get_settings () fn in other operators yoy may be able to figure out how to fix it if you want to get your feet wet with code.

patmo141, 2016-03-09:
Also my phone typing is terrible. Apologies for all the typos

Leeda611, 2016-03-09:
Thank you for your quick reply! I'll have a look and give it a try (first coding attempt for me..lol)

patmo141, 2016-03-09:
Ok, I was able to edit from the web browser.  See if it works now. I have no way of testing.

Leeda611, 2016-03-09:
I tried a couple of times, still getting the same error.. key "odc_public" not found.
I opened the scripting dialog to have a look..(all red ) here's the pic.
Appreciate your help! Thank you!
[image]

Leeda611, 2016-03-09:
ah... let me try this again... will keep you posted.. Lee :)

patmo141, 2016-03-09:
ok, I was going to ask if you downloaded, uninstalled, restarted blender and installed again

Leeda611, 2016-03-09:
ok... new error. Refractory?  ( thank you for waiting while I tried again with this)

[image]

patmo141, 2016-03-09:
Ah, different issue.  You need the loopstools addon enabled.  it comes with blender.

Leeda611, 2016-03-09:
Thank you for your time and help!  We have a splint! :)
[image]
http://screencast.com/t/cQpB1zmCFSN

patmo141, 2016-03-09:
I have started a separate issue for the loop tools dependency.  Closing this one.  Thanks for your detailed bug reports.

closed 2016-03-09 by patmo141

Leeda611, 2016-03-09:
Thank you for your expert help! 
Lee