patmo141/odc_public #11: Creating outline in for split error

debaetsr, 2016-03-14:
Hello, 

I receive following error while making a splint:
Blender version 2.77
Import STL -> create splint (linked) -> Mark Splint outline

Traceback (most recent call last):
  File "/home/ruben/.config/blender/2.77/scripts/addons/odc_public-master/splint.py", line 767, in modal
    nmode = FSM[self.mode](context, event)
  File "/home/ruben/.config/blender/2.77/scripts/addons/odc_public-master/splint.py", line 722, in modal_main
    self.crv.click_add_point(context, x,y)
  File "/home/ruben/.config/blender/2.77/scripts/addons/odc_public-master/curve.py", line 713, in click_add_point
    loc, no, face_ind = self.snap_ob.ray_cast(imx * ray_origin, imx * ray_target)
ValueError: too many values to unpack (expected 3)

location: <unknown location>:-1

Regards,
Ruben

patmo141, 2016-03-14:
Hi reubwn, this is due to API change from 2.76 to 2.77.  Try in 2.76 for now.  I'm not ready to make everything 2.77 yet

patmo141, 2017-03-29:
This has been fixed for Blender version 2.77 and greater.

closed 2017-03-29 by patmo141