patmo141/odc_public #14: Flexi Tooth

Guisan, 2016-04-01:
Hi Patrick!
Flexi Tooth don't work on Pontic Crown.

Thanks
Guido
[image]

patmo141, 2016-04-01:
Thanks guido,  you are right!  I had not considered the changes to the mesh cased by pontic.  This will take me some time to fix.

Best,
Patrick

patmo141, 2016-05-01:
the mappings are messed up for some teeth as well.

[image]

Guisan, 2016-05-01:
Will it take a long time to reorder them?

patmo141, 2016-05-01:
no it's just tedious

Guisan, 2016-05-15:
Could I help you? I can't write in Pyton language but, if it's an easy thing, I can try.

patmo141, 2016-05-31:
[image]

should be all taken care of.

closed 2016-05-31 by patmo141

rsrudy, 2016-05-31:
Hello Patrick ,  
sorry for my beginner question :
  to get this improvement on my ODC , and all the modifications that came previously , and that will come later , what is to be done ? 
Download them and activate them one by one ( they are only texts , not files , i know how to download a file to blender ,not a text  , any link that might help if i need to download them separately as texts scripts plse ).......... ,

 OR JUST I RE-DOWNLOAD THE WHOLE ZIP FILE again ? and all the improvements  would be already  include by you in the main  ODC zip file? )

Thank you

patmo141, 2016-05-31:
"OR JUST I RE-DOWNLOAD THE WHOLE ZIP FILE again ? and all the improvements would be already included by you in the main ODC zip file? "

Yest this is correct! You will need to uninstall the old open dental cad from user preferences, then restart blender, then install the new one from the .zip file.

Alternatively, an advanced way to do it is here
https://github.com/patmo141/odc_public/wiki/Install-Intermediate

Guisan, 2016-05-31:
Ok Patrick! It works but the processing is slow.
https://youtu.be/EiHjDcJ6U_c

patmo141, 2016-06-04:
Wow that is slow!  And when you use it on other teeth (that are not pontics) it is faster or is it slow for ALL teeth?

Guisan, 2016-06-04:
Slow only if you use on the pontics elements.

patmo141, 2016-06-04:
let's open a new issue for that #25

commit 8845ed99 (2016-05-31): improve flexi tooth for pontics

fix #14