patmo141/odc_public #15: Set mesal Set distal

Guisan, 2016-04-01:
Hi Patrik!
After selection of the mesial surface in edit mode, by pressing the key "Set mesial" nothing happens...or I am incapable! 
The same result for "Set distal".

Thanks
Guido

patmo141, 2016-04-01:
It's happening, it's just not obvious.

patmo141, 2016-04-01:
So I will try to think of a way to add better feedback so the user is reassured that it has been successful.  Thanks for all the reports, it is really helpful to improving and strengthening the software

Guisan, 2016-04-01:
It 'obvious that something escapes me!
[image]

patmo141, 2016-04-01:
Can you save that .blend file and share it via Dropbox or other way (you can upload to pasteall.org)  and I will look at it tonight when I get home.

Guisan, 2016-04-01:
[https://www.dropbox.com/s/5su4yu62mmw2qya/Test26.blend?dl=0](url)

Guisan, 2016-04-01:
https://www.dropbox.com/s/5su4yu62mmw2qya/Test26.blend?dl=0

patmo141, 2016-04-01:
thanks, I just sat down to start working

patmo141, 2016-04-02:
https://github.com/patmo141/odc_public/wiki/Set-Mesial-Set-Distal

patmo141, 2016-04-02:
Try those improvements!  See if that works to your satisfaction

Guisan, 2016-04-02:
Hi Patrik!
It works! I am very satisfied!
Thank you very much!

Guisan, 2016-04-04:
All right!
[image]
Thanks!

closed 2016-04-05 by patmo141