patmo141/odc_public #16: Missing intaglio for some abutments

airmoroz, 2016-04-26:
Hello,I like odc very much.It has a big perspectives .
I've started working with this program for a month ago.And now I have some troubles.such as:
I can't soldify bridge. Error- Missing intaglio for some abutments.
I've watched video tutorial "new splint guide temporary bridge" for many times,and have practiced to make splint with guides,but every attempt on soldify bridge stage:

Missing Intaglio for some abutments 
Traceback (most recent call last): 
File "C:\Users\user\AppData\Roaming\Blender Foundation\Blender\2.76\scripts\addons\odc_public-master\bridge.py", line 252, in execute 
new_me = ob.to_mesh(context.scene, True, 'PREVIEW') 
AttributeError: 'NoneType' object has no attribute 'to_mesh' 

location: <unknown location>:-1 

Traceback (most recent call last): 
File "C:\Users\user\AppData\Roaming\Blender Foundation\Blender\2.76\scripts\addons\odc_public-master\splint.py", line 32, in poll 
implants = odcutils.implant_selection(context) 
File "C:\Users\user\AppData\Roaming\Blender Foundation\Blender\2.76\scripts\addons\odc_public-master\odcutils.py", line 685, in implant_selection 
b = context.user_preferences.addons['odc_public'].preferences.behavior 
KeyError: 'bpy_prop_collection[key]: key "odc_public" not found' 

location: <unknown location>:-1 

Missing Intaglio for some abutments 
Traceback (most recent call last): 
File "C:\Users\user\AppData\Roaming\Blender Foundation\Blender\2.76\scripts\addons\odc_public-master\bridge.py", line 252, in execute 
new_me = ob.to_mesh(context.scene, True, 'PREVIEW') 
AttributeError: 'NoneType' object has no attribute 'to_mesh' 

location: <unknown location>:-1 

Traceback (most recent call last): 
File "C:\Users\user\AppData\Roaming\Blender Foundation\Blender\2.76\scripts\addons\odc_public-master\splint.py", line 32, in poll 
implants = odcutils.implant_selection(context) 
File "C:\Users\user\AppData\Roaming\Blender Foundation\Blender\2.76\scripts\addons\odc_public-master\odcutils.py", line 685, in implant_selection 
b = context.user_preferences.addons['odc_public'].preferences.behavior 
KeyError: 'bpy_prop_collection[key]: key "odc_public" not found' 

location: <unknown location>:-1 
------------------
Maybe because of it, i can't make inner cylinder ,guide cylinder etc
Need help,please.
Thanks.

patmo141, 2016-04-27:
please upload a .blend file to dropbox!  Thanks

patmo141, 2016-04-27:
most of the functions needed are still there.  I will post an updated tutorial on the WIKI as soon as possible.

[image]

airmoroz, 2016-04-27:
Hello.blend file in dropbox:
https://www.dropbox.com/s/90t0u5eq2pyg75g/missingintagliofsabuts.blend?dl=0
I look forward to see new tutorial on the wiki.
Thank You.

patmo141, 2016-04-27:
https://github.com/patmo141/odc_public/wiki/Temporary-Bridge-(Implant)

I've started working on the wiki page.  I'll download and explore your blend file tonight.  Best, Patrick.

patmo141, 2016-04-28:
I have updated the wiki (partially).  I am reviewing your file but have not found the reason for the error yet.  Great case though.  Huge plan.

airmoroz, 2016-04-28:
I haven't set 'insertion axis" for teeth in this project(i thought that was a sense of error),but this act haven't solved my problem.I have tried from the beginning more and more, but the error was the same.I've downloaded Your file with basic splint designed and made all from instructions,but i couldn't soldify bridge and "inner cylinders" etc.. for "finishing" splint. ((
ps: Thank You for new wiki guide.every time i am online -i check first Your github page and zohodiscussions forum(there are some interesting topics ,especially "volume data in the 3d view").
Thank You.

patmo141, 2016-04-29:
Ok,

it seems that when you do "pontic from crown" it does not update the project to know that the tooth is a pontic.  It is still looking for an intaglio, even though there is not one.  I should have this fixed ASAP!

patmo141, 2016-04-29:
-I fixed the restoration plan manually (eg, with code) so that the pontics were treated as pontics.
-I sculpted the distal of of the canine and the mesial of the premolar to make them intersect
-If you wanted 2 separate bridges (21x23 and 24x27) then it is best to just plan 2 separate bridges
 -I re-created the bridge because the original failed attempt to join the pontics altered part of the bridge margins
-I also re-formed the intaglios of 21 and 23 because somehow they got deformed a little.

Here you go!  I will push updated code so that you can complete the case on your own.
https://www.dropbox.com/s/l9i47x0hmsu7p9w/intaglios_fixed.blend?dl=0

patmo141, 2016-04-29:
[image]

commit 11e56971 (2016-04-29): fix #16
closed 2016-04-29 by patmo141