patmo141/odc_public #17: Enable addon error

amboleos, 2016-04-28:
When I tried to enable the addon, I got the this

[image]

Im using 2.73 blender version.

patmo141, 2016-04-28:
Currently this addon is based on 2.76, soon to be upgraded to 2.77.

last82, 2016-11-22:
Same error here, please update ;)

patmo141, 2016-11-22:
try with blender 2.77

slimane77, 2017-12-31:
hello,
i follow exactly the steps of easy install, but i can't found the dental addon on my macbook pro version installation.
please tell me what's the solution, i install all the version required but no one work.
best regards

andy-s-hub, 2020-08-31:
hallo,ich habe das addon ODC-public instaliert,und bekomme ich immer fehler.ich benutze blender 2.83.1

patmo141, 2020-08-31:
can you please provide a screen shot of the errors in the system console?

andy-s-hub, 2020-08-31:
[image]

andy-s-hub, 2020-08-31:
das kommt wenn ich set Master drücke

andy-s-hub, 2020-09-01:
hallo,ich habe blender2.77 instaliert,und leuft das was ich möchte.kronen und brücken modelieren.danke

andy-s-hub, 2020-09-01:
vielleicht könne Sie mir schreiben bis welchen kann ich aktualisieren,danke

andy-s-hub, 2020-09-01:
ob es mit blender 2.8x leuft

patmo141, 2020-09-01:
Hi andy-s-hub.  A first start us to remove "layer_management" from
odcutils.  Layers are no longer used in 2.8, then just regular API changes.

Best,
Patrick

andy-s-hub, 2020-09-01:
wenn Sie mir das zeigen könnten,wo die odcutils zufienden sind