patmo141/odc_public #18: Artificial Intaglio

Guisan, 2016-04-29:
Hi Patrick,
Unfortunately the carving does not follow the margin.

Thanks
Guido
[image]

patmo141, 2016-04-29:
Did you alter the crown shape after calculating artificial intaglio?  Or did this result display immediately.  Unfortunately it does not update live.  I thin you just need to recalculate artificial intaglio.

Guisan, 2016-04-29:
1) I altered the crown shape with lattice deform crown.
2) I calculated artificial intaglio.

patmo141, 2016-04-29:
Have you done "keep tooth shape" after the lattice deformation?

Guisan, 2016-04-29:
Yes, I did!

Guisan, 2016-05-01:
A short video to see what happens.
https://youtu.be/3B5PvXsadbw

Thanks

patmo141, 2016-05-01:
Thanks,  That is very strange!  I'm looking into it.

commit d59408f0 (2016-05-01): fix #18
closed 2016-05-01 by patmo141