patmo141/odc_public #19: Error in Splint making

seifeldin88, 2016-05-18:
Hello i am trying to make a splint, but after i set my points of outline and calculate bezier splint, it gives me an error as shown..
what could be the problem? please help
[image]

patmo141, 2016-07-26:
Hi,

This error is because you do not have the Looptools addon enabled.  I updated the wiki installation isntructions to include this.  Please see item 5 on this page

https://github.com/patmo141/odc_public/wiki/Install-Easy

This is actually related to issue #8 which I am trying to minimize the dependencies on other addones.

DrHugoAndres, 2016-08-27:
Hi, 
Get Started!The addon is in transition right now! I Instaled blender cad 2.77 e windows 10 
and I can not draw the line splint, no activate initiate_arch_curve. Whats appening!