patmo141/odc_public #21: Calculate Intaglio

Guisan, 2016-05-21:
Hi Patrick,
some problem with "Calculate Intaglio".
https://www.youtube.com/watch?v=KMMdHUZMxno

Thanks
Guido

patmo141, 2016-05-23:
Guido, can you un-hide the pseudomargin so I can see it?  Or post a .blend file?  I think it must be something with that small concavity and my algorithm.

re:  Angle Cervical Convergence, try -15degrees and let's see what happens.  There may be a normals problem

Guisan, 2016-05-26:
Hi Patrick, after so many tests I have achieved positive results! The intaglio mesh is a bit messy as well as the cervical area and the margin.
https://www.dropbox.com/s/nia20vgke5v93j5/IntaglioTest.blend?dl=0

Guisan, 2016-05-28:
This is the file where intaglio and angle cervical convergence have problems.
https://www.dropbox.com/s/7gnpt0r8zsk8k9l/IntaglioKO.blend?dl=0

Thank you Patrick
Guido

patmo141, 2016-06-22:
I have almost finished fixing the intaglio problems!  A complete and total rewrite of the method!  Will update tonight.

Guisan, 2016-06-23:
Hi Patrick, I can't wait to try it!

patmo141, 2016-06-24:
[image]

closed 2016-06-24 by patmo141

Guisan, 2016-06-29:
What a good job! It works perfectly and it's possible to obtain a tidy mesh.
Thank you,
Guido

commit ee7f71ed (2016-06-24): update addon to 2.77 with backwards compatability for 2.76

fix #21