patmo141/odc_public #23: no success without "Mesh relax"  and "Mesh loop tools"

rsrudy, 2016-05-21:
This is for any Blender beginner such as myself  . Another thing that happened with me is that i could not execute the "calculate intaglio" , it always gave error message .  Until i discovered that i had to go to   USER PREFERENCES  , click ADD-ONS .and there in the search window (upper left), write the word "RELAX" , you will get the sentence MESH:RELAX : activate it by checking the box ,and then click SAVE USER SETTINGS (lower left)....

.Again in search window , write the word "LOOP", you will get the sentence MESH:LOOP RELAX :activate it also by checking the box , and as said above ,click save user settings ........now you have both these add-ons active , and that fixed the issue of "calculate intaglio".............

patmo141, 2016-05-23:
Thanks for your bug reports!  I will update the wiki installation instructions to include enabling those 2 addons, also this is related to issue #8 minimizing dependencies.  I will close this issue when I have update the wiki.

patmo141, 2016-05-25:
I have updated the wiki pages to reflect enabling those add-ons in the setup instructions

closed 2016-05-25 by patmo141