patmo141/odc_public #24: ...cannot see "PlanSelectedUnitsAsBridge"...

rsrudy, 2016-05-29:
Hello Patrick , 
When i started from scratch , i installed the "odc_public-master.zip"file(68.7Mb), to blender and got it functional .I can use the "teeth:Tooth Restoration"commands normally. But the "bridges : "Bridge Restorations", i can only use the first three(DrawArch,SetTeeth,KeepArch).After  i draw arch, i set the teeth , i keep arch , whatever i do ( try ctrl/join teeth, or else ) , i cannot get the other commands to illuminate so that i can use them , "PlanSelectedUnitsAsBridge""MakeBridgeShells" etc.. was not able to get them and use them ....what i am missing here ?...should i download any additional add on files AFTER downloading the main initial ODCzip file ?  Because i did not do any additional downloads from your page that i attach also a pic here .
Thank you very much .
[image]
[image]

patmo141, 2016-05-29:
They are not active because you don't have any restorations in the scene.

As I understand, you have succcesfully planned an arch by following these instructions
https://github.com/patmo141/odc_public/wiki/Arch-Planning

Once you have actual teeth in the scene, you need to select them (with right mouse).  Then you will be able to create a bridge from those teeth.

Please see the "Plan a Bridge Restoration" in this advanced tutorial
https://github.com/patmo141/odc_public/wiki/Temporary-Bridge-(Implant)

Here the image
[image](https://camo.githubusercontent.com/f13b9e36dc1bf5e9f8e2aff5c8630bd1c4247710/68747470733a2f2f646c2e64726f70626f7875736572636f6e74656e742e636f6d2f752f323538363438322f6f64635f7075626c69635f646f63756d656e746174696f6e2f54656d70253230427269646765253230496d706c616e742f73656c65637465645f756e6974735f746f5f6272696467652e706e67)

rsrudy, 2016-05-29:
Thank you fro your reply .
(model here is only for demo purpose)
yes indeed,
I believe i already did as you wrote here , the thing is that even when i click the teeth , i do not get any thing , the tooth number is even not there . I guess am still not doing it right but cannot figure what :) ....
[image]
[image]
[image]

Anyway , i will wait for any future public video with " a to z steps" , when you will have the time to do it . Thanks a lot for your time . Best regards

patmo141, 2016-05-29:
When you did the arch plan, did you have the "link" option checked?  The yellow box here.  If you did, then this may be a real bug.  Please post the blend file on dropbox link.

[image]

rsrudy, 2016-05-30:
ooook , it worked now ...THANK YOU  :)

closed 2016-05-30 by rsrudy