patmo141/odc_public #26: automatic bridging not available

rsrudy, 2016-06-07:
Hello Patrick ,

F
[image]
[image]

or some reason i cannot see the automatic bridging function , and after bridge design i am "connecting " teeth manually in edit mode ... any idea why i do not have this bridging function , please advise . Thank you and very best regards .

patmo141, 2016-06-27:
For now I have disabled the method which bridges "edge loops" between the abutments and pontics because I was having a difficult time getting consistent results.  I disabled it in favor of a Boolean based bridging operation.  I can however re-enable it, all i did was remove it from the menu's so the code is still there.  I will try and update the method for better results and turn it back on.

Sincerely
Patrick

rsrudy, 2016-06-29:
Hello Patrick ,  ook , thanks a lot ! Very best regards

rsrudy, 2016-06-29:
...and  how can we use the " boolean based bridging operation" ?...