patmo141/odc_public #27: Can find information needed on wiki

franchise1003, 2016-06-13:
hello patmo141
I'm searching for a tutorial about split single tooth one by one to make a arch with separated teeth.
Is this function available right now? I can use make margin to make a circle of one teeth,and then I have no idea what to do......
Can I have my own request about this project?
I have a bunch of idea about the teeth, thanks!

patmo141, 2016-06-24:
For cutting/segmenting teeth on the model, the "cut_mesh" addon (which is still beta status) can be used.  For ideas you may have, please feel free to post them on the forum https://www.zohodiscussions.com/blenderdental#home

patmo141, 2016-06-25:
https://zohodiscussions.com/blenderdental#Topic/271312000000010001

github.com/patmo141/cut_mesh

patmo141, 2016-07-26:
Hi Franchise, were you able to find the information you needed?

closed 2016-07-26 by patmo141