patmo141/odc_public #29: Trying to use Polyline Trim Mesh

AxlWRose, 2016-08-08:
It doesn't work...
After pressing C button - lots of errors:

**_**Traceback (most recent call last):
  File "C:\Blender 3D\Scripts\addons\odc_public\splint.py", line 844, in plyknife_draw_callback
    self.knife.draw(context)
  File "c:\Blender 3D\2.77\scripts\modules\bpy_types.py", line 598, in __getattribute__
    properties = StructRNA.path_resolve(self, "properties")
ReferenceError: StructRNA of type OPENDENTAL_OT_mesh_trim_polyline has been removed cutting!

Traceback (most recent call last):
  File "C:\Blender 3D\Scripts\addons\odc_public\splint.py", line 1017, in modal
    nmode = FSM[self.mode](context, event)
  File "C:\Blender 3D\Scripts\addons\odc_public\splint.py", line 934, in modal_main
    self.knife.make_cut()
  File "C:\Blender 3D\Scripts\addons\odc_public\curve.py", line 289, in make_cut
    self.snap_poly_line()
  File "C:\Blender 3D\Scripts\addons\odc_public\curve.py", line 242, in snap_poly_line
    loc, no, ind, d = self.bvh.find(imx * v)
AttributeError: 'BVHTree' object has no attribute 'find'

location: <unknown location>:-1**_**

What am i doing wrong?
Windows 7 x64
Trying in both Blender 2.76 x64 and new script in Blender 2.77 - same error

patmo141, 2016-08-08:
Hi,

You are not doing anything wrong.  I started developing the cutting tools in open dental before realizing it needed to be it's own separate project because it is a very complex problem (for me). I need to remove this function from ODC.  Please see github.com/patmo141/cut_mesh for the newer, faster and almost complete cutting tools.

AxlWRose, 2016-08-08:
Thanks a lot for your answer
I use cut_mesh as basic tool
But when i saw your videos - everything works ok. So I decade to ask - may be my configuration (Win x64 + Blender) is not good :) in compare with Debian or Mac

So, thank you for you HUGE JOB!!!

patmo141, 2016-08-08:
You are very welcome.

closed 2016-08-08 by patmo141