patmo141/odc_public #35: Implant problems!

DrHugoAndres, 2016-10-11:
Hi, good morning!!!
The cylinder guide does not match its axis with the implant and top edge to apex of implant don't receive negative values! What's appening?
[image]
[image]

DICOM images would be better for references to nerves and bones. would have to be split into three or four windows!

patmo141, 2016-10-11:
Please upload thus .blend file.  Cerainly the cylinder should match.  Either this is new bug from API change or perhaps you accidentally translated it.  Sometimes 'right click and drag' will accidentally move an object.  I will add protection to prevent this from happening in next update.

DrHugoAndres, 2016-10-19:
Hi, Good afternoon! Sorry, I'm dentist busy because Merry Crhismas!
I uninstall and install blender and ODC. Please, this is a new bug!
I would like to implement in practice daily clinical and no possible yet!
I have acess into a CAD/CAM systems.
I'd like to have a best practice for get started in programming in blender. May you do help me? Links?

DrHugoAndres, 2016-10-21:
Hi, Patrick!

I managed to learn how to use your addon. it is difficult and requires a lot of abstract knowledge. in the last month I tried to simulate a very complex treatment. So my failure to date. I'd like to congratulate him for his immense work and attention.

The Best Regards

closed 2016-10-21 by DrHugoAndres