patmo141/odc_public #40: Occlusion with Set Teeth on Curve

patmo141, 2017-01-20:
The operator Set Teeth on Curve currently re-scales the upper and lower arches to be exactly the same length, which in clinical reality is not the case.  To better be able to simulate real/hamonious occlusal schemes, a smarter calculation will be better.

[image]

Guisan, 2017-01-20:
Very good!
Is it also possible to tilt the incisal zone of the canines and the incisors toward the vestibule?
[image]

patmo141, 2017-01-21:
yes but that will have to come from the design of the tooth library.  I will post more information on Wiki once I have it all sorted out.  Marcos Graf has been working on adjusting the tooth library so that the tip and torque are stored in the local coordinates of the tooth.

patmo141, 2017-01-25:
[image]

One problem is that the bounding box width does not represent the width of the tooth along the curve, especially on the lingual contacts of the upper anteriors.

patmo141, 2017-01-25:
[image]

This method works better by taking a cross section at the occlusal point and using that to measure

patmo141, 2017-01-26:
[image]