patmo141/odc_public #42: ODC Tooth Library

Guisan, 2017-01-24:
I noticed that in the ODC Tooth Library the normals of the left sector (21-27 31-37) are to be recalculated.
Also the topology of the mesial and distal connectors can be improved so as to create equivalent areas of vertex groups.

patmo141, 2017-01-24:
1.  The normals are a result of an old Blender bug.  When mirroring the object, or flipping normals, it would destroy the Multires Sculping.  I think this bug has now been fixed, so we can "flip" the normals on the left teeth. Please do not do this by hand, I will write a script which complete's the task in seconds.
Also, I will need to make changes to some of the methods because some compensate for the normals.

2.  Changing the topology will most likely destroy the sculpting anatomy.  Also, some of the operators rely on the current topolgy (flexi tooth, seat to margin etc). It is a good goal, but should probably be undertaken soon because the further we go with current system, the more work will be needed later.

patmo141, 2017-03-27:
The normals in the standard ODC library are correct, and the sculpting behavior is consistent from the right to the left side.
[image]

patmo141, 2017-03-29:
In the Brenes library, there is some confusing behavior to me.  The normals are correct on all the base meshes, but the sculpt behavior is reversed on the left side objects.

[image]

closed 2017-03-30 by patmo141