patmo141/odc_public #47: Library

DrHugoAndres, 2017-11-19:
I'm in the final stages of the Breath Sleep Splint, i'm validating protusion gauge! I would like to make a contribution in your repository.
I need to know how to create/increase tooth libraries, implants, abutmans and  to focus on increasing ODC functionality. I did not understand how you managed to divide your teeth into mesial, lingual, vestibular, is there a code for this? Can you show me?

patmo141, 2017-11-19:
Open up one of the tooth library .blend files in the "data" folder  
[image]

patmo141, 2017-11-19:
1)  First thing to notice is that the local orientation of each tooth has the following properies
-The Z axis points toward the Occlusal
-The X axis poinrts toward the Mesial
-This means that the Y axis will point in different directions for different quadrants (Toward the Facial for quadrants 1 and 4, and toward the palatal for quadrants 2 and 3.

[image]

patmo141, 2017-11-19:
2.  The teeth must be named with the FDI system.  16 is the upper right molar.
[image]

patmo141, 2017-11-19:
If you follow those first 2, then you library will work for tooth placement, waxup, tooth to curve and some other functions, but it will not work for  Crown/Bridge or Flexi Tooth.

patmo141, 2017-11-19:
Topology and Vertex Groups

For you teeth to work with the crown and bridge tools (like seat to margin), the teeth have to be very stricly designed.  For this you will need to make a copy of the ODC_toot_libaray and then modify those teeth.

1.  Vertex Groups:
[image]

2. Multires Modifier on level 3.  So you will start with the basic tooth library, and then modify/sculpt/snap the higher multires data levels to match your desired tooth shape.
[image]

DrHugoAndres, 2017-11-20:
Can I make a mirror modifier to sculpt only half of them? or for scan 3d retopolize and mirror the teeth?

DrHugoAndres, 2017-11-20:
Its more easy for build the library!

patmo141, 2017-11-20:
Yes, only sculpt half, and we mirror to get the others.  Don't use mirror modifier .  Yes you can retopologize scanned teeth by using shrink-wrap to adapt the original odc teeth to the scan data.

DrHugoAndres, 2017-11-21:
In the right (Toward the Facial for quadrants 1 and 3) I sculpt them and the left ( toward the palatal for quadrants 2 and 4) I scale lower posterior teeth.
[odc_tooth_library_sculpted.zip](https://github.com/patmo141/odc_public/files/1492927/odc_tooth_library_sculpted.zip)

DrHugoAndres, 2017-12-07:
[odc_tooth_library_no_sculpted.zip](https://github.com/patmo141/odc_public/files/1540766/odc_tooth_library_no_sculpted.zip)
I don't have a free time now.
We should do a Denture library for manufacturing the tooth and gengival
[odc_tooth_library_no_sculpted.zip](https://github.com/patmo141/odc_public/files/1540771/odc_tooth_library_no_sculpted.zip)