patmo141/odc_public #53: Update to blender 2.8?

ephestione, 2019-10-09:
Just putting it here for good measure, if you even plan to spend time on this tool again :)

patmo141, 2019-11-09:
would you like to help or could you help addon developers who would like to help? I am going to invest in some updating to odc_public as time allows.

ephestione, 2019-11-09:
I have no knowledge whatsoever of python, I am only somewhat fluent in PHP :disappointed:

aak277, 2019-11-10:
> would you like to help or could you help addon developers who would like to help? I am going to invest in some updating to odc_public as time allows.

i would like to help addon developers who would like to help. i am no developer myself however i am a dentist who dabbles in source code.

subhamsamir, 2020-04-25:
I am interested, I am a noob in python but a quick learner, Facing the same issues. I can help in any way If possible

patmo141, 2020-04-25:
yes this is happening!  I have several people tinkering with it.  I'm putting a little effort in.  I have to focus on my commercial products but as I'm updating those, I am putting some time in on ODC and vice versa.  @subhamsamir  go ahead and fork the b_280 branch.  We are about to push some updates to it in a couple of days.