patmo141/odc_public #65: Help!! Questions in your Blender video

wangmyde, 2020-05-06:
Hi, I met the same problem and finally found your answer. I have downloaded your code and successfully run your test code. But I  can't do the steps of selecting a point on the 3D model and the image. I mean how you can do from 2:20 in the video. In my Blender When I clicked the left button of my mouse, there was nothing on the model or the image or coordinate. I don't understand what you mentioned that "selection up points in the 3D view with ray casting... " Because I am new to Blender, can you detailed tell me how you can select the gold/yellow point in your video from 2:20? Thanks very much.

patmo141, 2020-05-06:
can you link the video here too please so I'm sure what you are referencing.  Hard to keep track of everything.

wangmyde, 2020-05-06:
Hi, thanks very much for your reply.

Here is the link: https://www.youtube.com/watch?v=Edk9uA5Yl5k&feature=youtu.be from 2:20.

I don’t understand what it is and how to do it in Blender to select those gold/yellow points. When I tried left clicked my mouse, nothing happened. Is there any button or mode should I turn on? Could you detailed tell me the step to do it? Thanks.

Mengyue

wangmyde, 2020-05-06:
> can you link the video here too please so I'm sure what you are referencing. Hard to keep track of everything.

Hi, thanks very much for your reply. 

Here is the link: https://www.youtube.com/watch?v=Edk9uA5Yl5k&feature=youtu.be from 2:20.

I don’t understand what it is and how to do it in Blender to select those gold/yellow points. When I tried left clicked my mouse, nothing happened. Is there any button or mode should I turn on? Could you detailed tell me the step to do it? Thanks.

Mengyue

wangmyde, 2020-05-06:
> can you link the video here too please so I'm sure what you are referencing. Hard to keep track of everything.

Hi, could you also let me know in 2:17 in the video how the panel “register image for object” shows up? I mean whether you know the shortcut to show this panel? I don’t know where it comes. 
[image]
 

And after ran the code, I was not able to let the panel show up.
```
import sys

sys.path.append(r'I:\Blender_old\blender-2.78-windows64\2.78\scripts\modules\image_object_registration.py')

import image_object_registration

image_object_registration.register()
```
Thanks.

wangmyde, 2020-05-06:
> can you link the video here too please so I'm sure what you are referencing. Hard to keep track of everything.

Thanks very much for your video. I have got the solution on how to do it. I missed the step of showing the panel so that I can't do the next step.  Thanks very much for your help. You are so nice.