patmo141/odc_public #66: New question for Blender video

wangmyde, 2020-05-06:
Hi, I followed your steps in your video, but I didn't get the expected result. 
https://www.youtube.com/watch?v=Edk9uA5Yl5k&feature=youtu.be

I am not sure whether my steps were correct because I can't see what you typed on your keyboard. I have some questions w.r.t.  your operation is:
1.  In your video from 8:23 to 8:25, which key did you type to step out the "registration image to object" process? I asked it because  I typed 'ESC' on the keyboard to be out of the process.  But it seems that the program can catch nothing with this operation.

2. I would like to know the whole process of using your program. Please correct the process I describe below:

1.  Open the blend file you upload
2. import image_object_registration module
3. deactivate the 'test camera' (I'm not sure whether it is necessary to do this step)
3. run 'Run me!' script
```
import sys

sys.path.append(r'I:\Blender_old\blender-2.78-windows64\2.78\scripts\modules\image_object_registration.py')

import image_object_registration
image_object_registration.register()
````
4. call out the panel of "registration image to object"
5. select 6 pairs of points in images and 3D model
6. Here I DON'T KNOW what to input to be out of the selecting points process, but I press 'Esc' on keyboard.
6. active the 'test camera'  (I'm not sure whether it is necessary to do this step)
7. run the script 'Camera matrix test'
```
import bpy
import image_object_registration as IOR
C = bpy.context

#these functions are useful!!
#get_blender_camera_from_3x4_P
#get_calibration_matrix_K_from_blender
#get_3x4_RT_matrix_from_blender
#get_3x4_P_matrix_from_blender

if C.object.type != 'CAMERA':
    print('not a camera')
    
else:

    P, K, RT = IOR.get_3x4_P_matrix_from_blender(C.object)

    print(P)

    IOR.get_blender_camera_from_3x4_P(P, 1)
```
8. activate the 'CamFrom3x4PObj' camera
9. view through the 'CamFrom3x4PObj' camera, then I would have seen the align image of 3D model as same as the 2D image. But nothing changed.  Here is the images:
[image]

Could you let me know where I am wrong? I have taken one night to explore your program.  I would be really grateful if you could correct me. Thanks.

patmo141, 2020-05-07:
>    1. call out the panel of "registration image to object"
>    2. select 6 pairs of points in images and 3D model
>    3. Here I DON'T KNOW what to input to be out of the selecting points
>    4. active the 'test camera' (I'm not sure whether it is necessary to
>    5. run the script 'Camera matrix test'
>    1. activate the 'CamFrom3x4PObj' camera
>    2. view through the 'CamFrom3x4PObj' camera, then I would have seen

I am glad you were able to get the code running!  The results I could get
were never perfect.  I actually made this video to ask for help from other
programmers since my results were not what I wanted.  I have not been able
to advance beyond the results you have.  With extremely careful placement
of your 3d and image points it can be decent.

patmo141, 2020-05-07:
Regarding between #5 and #6 in your question

#5  run the operator from the spacebar menu and select 6 (or more) corresponding points
#5a   press "M" on the keyboard to generate the matrix
#6  press "Enter" to confirm or "Esc" to get out of the points operator