patmo141/odc_public #71: Update the Wiki for 2.8?

apankow1234, 2020-07-13:
We are using Blender 2.8 and we have no idea how to get started. We already had to fix the root __init__.py file for "user_preferences" to "preferences" on line 78. But we're still not seeing any menus. Please assist. Thank you.

patmo141, 2020-07-13:
Hi, make sure you are checking out the most recent b_280 branch.  Not all of the functions have been ported over.   Some model work, some splint/shell functions and some new contributions by a few developers/

patmo141, 2020-07-13:
[image]
Actually we merged the b2_80 branch into master 11 days ago.  I just checked with latest b2.83.2 build everything is behaving as expected.

apankow1234, 2020-07-13:
Success! Our download of the master branch as older than that. Thank you!

closed 2020-07-13 by apankow1234

patmo141, 2020-07-13:
You mentioned that you made changes in __init__.py.  That's adventurous for new users.  Can we rope you into tinkering and making pull requests when you find things that can be fixed?

apankow1234, 2020-07-13:
I'm not a pro with GitHub/GitLab/social coding since I've taught myself but certainly. I have over a decade of experience in 3D and code (MFA in Animation and VFX).

apankow1234, 2020-07-13:
Do you guys have an UML breakdown or are you shooting from the hip?

patmo141, 2020-07-13:
Hip.  This project started as self taught python scripting.  Over time there is more OOP.  Code quality ranges from awful to excellent across the project.

apankow1234, 2020-07-13:
Understood! The lab I work for is definitely infatuated with the tool and wants to use it into the future. I'll keep my eyes peeled.

patmo141, 2020-07-13:
feel free to contact me directly patrick dot moore dot bu at gmail