patmo141/odc_public #74: tooth restorations problem

Anna486, 2020-10-04:
I wanted to check how ODC tooth restorations works on Blender 2.83, but even Crown Helper Wizard returns errors. What's wrong?
[image]
[image]
[image]
[image]

patmo141, 2020-10-04:
While the panels have been updated, a lot of the functions have not for the tooth restorations part of the addon.

Tixoserg, 2020-11-25:
Can you please tell me if it will be possible to add implants, bushings, or drills in the future, or is it just for information or developers? Thanks.

okkarkyaw122, 2023-05-08:
I am trying to make a single crown out of this. Can you post a video tutorial?